RockRaidersInc/Mechanical #50: Determine initial requirements of a communications mast.

J-Corbin, 2019-06-14:


J-Corbin, 2019-06-24:
David, assigning you. I don't know what we need on a mast or how tall it needs to be. Providing me some general requirements would be nice.

daweim0, 2019-06-25:
I think it should be at tall as practical for the antennas, that part doesn't have to be super rigid. I assume taller is better but I don't know how tall we need. Maybe just tall enough so that the antennas are a little higher than the arm? That would be a good question for Connor or Nisha.

 I would like to put a camera and GPS on the comms mast at least two feet above the body of the rover. The camera should be looking forward. The part of the mast holding the camera up does need to be rigid.

J-Corbin, 2019-07-01:
What's the relation between antenna height and signal strength going to be? Unless signal strength goes up a lot faster than antenna height, I'm wary of making something incredibly tall. Suppose just hypothetically that mounting the antenna at 4 feet high gives us a +20 yard range and mounting it at 5 feet gives us a +30 yard range. To me, that difference in range isn't enough to justify the necessary infrastructure required to mount it at the extra foot higher. 

The arm needs to be on the front of the rover, so where do you intend for the mast to go? Also, the entire mast would need to be rigid. Having it rigid only up to the camera and then less rigid after that to hold the antennas isn't a good idea. The antennas will experience a lot of motion, especially being higher up. If the part of the mast they're attached to isn't rigid, it's going to move too, which is going to put extra dynamic forces on the rigid part of the mast, causing it to wobble more also and additionally require stronger mounting at the interface between the two halves of the mast.

daweim0, 2019-07-03:
What if we start with a 2 foot mast, then worry about making it higher if we need more range later

J-Corbin, 2019-07-03:
What about just getting longer antennas for the radio on the rover? I haven't seen a lot of URC teams that use ZED cameras mounting them up high either

daweim0, 2019-07-03:
The team that won autonomous traversal in 2018 had their zed camera mounted higher up. I think teams targeting autonomous traversal generally do have a camera mounted somewhere high. We definitely could look into longer antennas

J-Corbin, 2019-07-05:
If we could get longer antennas so that we don't need to mount the radio on the end of the mast, I would really like that. The ZED and GPS don't weigh much, so making a stable mast for those is easy. It becomes more complicated if I also need to support the radio at a point higher than the ZED and GPS.

daweim0, 2019-07-05:
I don't think it matters if the radios are higher or lower than the zed and GPS. It's just good to have both high up.ask Connor to look into antenna extension cables, I know next to nothing about antennas.

closed 2019-07-09 by J-Corbin